Kl0ibi/esp32_hackingtool #7: Problem with flashing

Jirka5091, 2023-02-25:
After I flash the code as instructed it looks like there are no errors, but the screen on esp is black and nothing is happening. Btw I have the 4 mb one

I will be glad for any help.

This is the serial output:
`19:03:42.063 -> [0;32mI (32) boot_comm: chip revision: 3, min. bootloader chip revision: 0[0m
19:03:42.063 -> [0;32mI (39) boot.esp32: SPI Speed      : 80MHz[0m
19:03:42.063 -> [0;32mI (44) boot.esp32: SPI Mode       : DIO[0m
19:03:42.063 -> [0;32mI (49) boot.esp32: SPI Flash Size : 16MB[0m
19:03:42.063 -> [0;32mI (53) boot: Enabling RNG early entropy source...[0m
19:03:42.063 -> [0;32mI (59) boot: Partition Table:[0m
19:03:42.063 -> [0;32mI (62) boot: ## Label            Usage          Type ST Offset   Length[0m
19:03:42.096 -> [0;32mI (69) boot:  0 nvs              WiFi data        01 02 00010000 00010000[0m
19:03:42.096 -> [0;32mI (77) boot:  1 factory          factory app      00 00 00020000 00e00000[0m
19:03:42.096 -> [0;32mI (84) boot: End of partition table[0m
19:03:42.096 -> [0;32mI (89) boot_comm: chip revision: 3, min. application chip revision: 0[0m
19:03:42.096 -> [0;32mI (96) esp_image: segment 0: paddr=00020020 vaddr=3f400020 size=106e58h (1076824) map[0m
19:03:42.458 -> [0;32mI (430) esp_image: segment 1: paddr=00126e80 vaddr=3ffb0000 size=043a0h ( 17312) load[0m
19:03:42.458 -> [0;32mI (436) esp_image: segment 2: paddr=0012b228 vaddr=40080000 size=04df0h ( 19952) load[0m
19:03:42.458 -> [0;32mI (443) esp_image: segment 3: paddr=00130020 vaddr=40`

Kl0ibi, 2023-03-05:
hey, the partition table specifies a flash size of 16MB since you have a 4MB flash storage this will not work. with all the websites that are included in the flash memory you will not get below 4MB even if you change the partition table. should I make you a "light" version with less websites (captive portals) or is this explanation enough for you to implement it yourself?

Jirka5091, 2023-03-11:
In fact I don't think that I need the website at all.

Jirka5091, 2023-03-11:
can I just remove this portion of code? 
`void htool_api_start_captive_portal(uint8_t cp_index) {
    captive_portal_task_args.is_evil_twin = false;
    captive_portal_task_args.cp_index = cp_index;
    htool_wifi_captive_portal_start();
}`

kubbur, 2023-05-03:
> can I just remove this portion of code? In htool_api.c
> 
> ```
> `void htool_api_start_captive_portal(uint8_t cp_index) {
>     captive_portal_task_args.is_evil_twin = false;
>     captive_portal_task_args.cp_index = cp_index;
>     htool_wifi_captive_portal_start();
> }`
> ```

did you get it to work? i have the 4mb version also

Jirka5091, 2023-05-03:
> 

Sadly no, I kinda gave up and started doing my version

Kl0ibi, 2023-12-24:
Hey, my apologies for the delayed reply, however, I'm excited to share some great news with you. I've successfully developed a Terminal version of the hacking tool. It's now compatible with any ESP device that has at least 4MB of flash memory, ensuring all features remain fully functional and accessible. Looking forward to your feedback!

closed 2023-12-28 by Kl0ibi