Kl0ibi/esp32_hackingtool #9: lite version for 4MB devices without screen

That-Dude, 2023-06-30:
First of all I just want to say how incredible this project is, I have it working on two 16MB TTGO boards and it has been super useful in pen-testing work, replacing most of my Rpi implementations that perform similar tasks.

it's a niche request but I would love to have a stripped down version that works on 4MB esp32 boards that don't have a screen (the super cheap ubiquitous dev boards that run for hours on a battery).

I only need the fake hotspot functionality and a method to retrieve any saved username/password data, maybe via a separate URL on the device  /showmewhatyougot

Kl0ibi, 2023-12-24:
Hey, thank you for your feedback appreciate it! My apologies for the delayed reply, however, I'm excited to share some great news with you. I've successfully developed a Terminal version of the hacking tool. It's now compatible with any ESP device that has at least 4MB of flash memory, ensuring all features remain fully functional and accessible. Looking forward to your feedback!

That-Dude, 2023-12-25:
Someone has been busy! This looks very interesting.

Does this need to be connected to a device via a serial interface or can i control it over Bluetooth?

I was hoping to deploy an array of these on generic esp32 devices, allow them to collect fake hotspot passwords, then retrieve the data later via Bluetooth (or dream scenario, espnow using a master controller). I appreciate that is asking a lot 😀

Kl0ibi, 2023-12-28:
Hi, thank you!
Yes you need to be connected with a serial interface or you could use a rs232 to bluetooth module (HC-05 / HC-06) with this you archive a serial interface over bluetooth.

Regarding your feature request, which is quite interesting. Implementing it might be complex, primarily because the ESP32 has only one module that handles both Bluetooth and Wi-Fi. Also it particularly challenging to set up a station and ESP-NOW simultaneously. 

...But I will give my best!

That-Dude, 2023-12-28:
I would be super happy with a built in Bluetooth solution for remote control / receive captured data. I dont find Bluetooth is useful for pen-testing so it's available. 

I just read up on espnow and you're correct, it really wants to be the only device on the wifi transceiver. I wonder if it would make sense to just use 2 esp32 boards and dedicate one to espnow, like this guy did

https://youtube.com/playlist?list=RDCMUCtfYdcn8F8wfRA2BXp2FPtg&playnext=1&si=AUwOwTdI6KKrbYaZ

arrase, 2024-07-11:
[Here](https://github.com/arrase/esp32_hackingtool) is my "slim version"  for tiny devices without screen:

```sh
Used static DRAM:   56972 bytes (  67608 remain, 45.7% used)
      .data size:   22164 bytes
      .bss  size:   34808 bytes
Used static IRAM:  123826 bytes (   7246 remain, 94.5% used)
      .text size:  122799 bytes
   .vectors size:    1027 bytes
Used Flash size : 2089679 bytes
           .text:  994343 bytes
         .rodata: 1095080 bytes
Total image size: 2235669 bytes (.bin may be padded larger)
```

M6ct0night, 2024-08-08:
We want to extract the files from the compilation, change its language, and add some new features. How can we open this compilation file, and which programs can we use to edit this file?