Kl0ibi/esp32_hackingtool #13: Will the project ever be updatet?

Playedswag, 2023-12-22:
Hey i love this project, and the tools are really fun. But will there be ever new features or is it discontinued`?

Kl0ibi, 2023-12-22:
Hey, in the coming days, I'm releasing a new update! This update introduces a command line interface, which means you won't need a display to operate it. Moreover, I've ensured compatibility with all ESP32 devices equipped with a  minimum of 4MB of flash memory. Additionally, I've incorporated some new BLE features and a range of useful general network tools.

I apologise for the sporadic updates; balancing this project with my very limited free time is quite a challenge.

closed 2023-12-22 by Kl0ibi

Playedswag, 2023-12-22:
Thanks man! I didnt mean to stress you, i just wanted to know :D

Playedswag, 2023-12-22:
And i have a feature request, if it would be possible, it would be pretty cool to have a option to choose a pin for a Ir sensor to control a tv.

Kl0ibi, 2023-12-24:
Great idea! I'll add it to my todo list. P.S. The previously announced update has now been released!