Kl0ibi/esp32_hackingtool #15: Bug/ New Idea

Dulikopter, 2023-12-26:
Hello i noticed some bugs i think:
- On captive Portal, the wifi name of google is "Google Free Wifi Test" i think the "test" should not be there
- On captive Portal of Facebook, you can just sign in with email, it should be email or phone number or not?
-deauth is not working on my home wifi
-eviltwin same
 New ideas:
 -Ddos tool
 (-PMKID capture
-WPA/WPA2 handshake capture and parsing
-Formatting captured traffic into PCAP format
-Parsing captured handshakes into HCCAPX file ready to be cracked by Hashcat
-Passive handshake sniffing
-Easily extensible framework for new attacks implementations
-Management AP for easy configuration on the go using smartphone for example)These are the feature of [ESP32-Wi-Fi-Penetration-Tool](https://github.com/0x1381/ESP32-Wi-Fi-Penetration-Tool)
-Spamming bluetooth device like airpods to smartphones, that someone cant use his phone because of pop ups of connecting with airpods

Dulikopter, 2023-12-26:
Maybe more bugs:
- Feature from beacon 2) - SSID random MAC, (3) - SSID same MAC are not working I think
Thanks in advance, and happy xmas!

Kl0ibi, 2023-12-28:
Hi, thank you for your valuable feedback!

Regarding the captive portal for Google's Wi-Fi, you've pointed out that its SSID is "Google Free Wifi Test." I agree, the word "test" seems unnecessary here. I'll look into adjusting that.

For the Facebook captive portal, you mentioned that it only allows sign-in via email. You're right, it would be more user-friendly to offer the option of signing in with either an email or a phone number. I'll make this change.

Regarding the issue with deauth not working on your home Wi-Fi for the eviltwin, it's intriguing as this is the first report of such an issue. Could it be that your router is blocking deauth messages? This might be worth investigating.

I appreciate your suggestion about adding DDoS tools. Indeed, I've already planned to incorporate some of these tools.

As for the PMKID capture technique, I understand your concern. This method requires a password list for reference, and its effectiveness can be questionable. Hence, I'm not particularly inclined towards this approach.

You mentioned the feature of spamming Bluetooth devices, like AirPods, to smartphones, causing persistent pop-up connection requests. This feature is available and can be accessed within the BLE spoof menu.

Finally, about the features from beacon 2) - SSID random MAC, and 3) - SSID same MAC. These features primarily impact devices that support mesh switching and determine mesh connections based on SSID/MAC. However, many devices tend to filter out duplicate SSIDs from their scan results.

M6ct0night, 2024-08-08:
We want to extract the files from the compilation, change its language, and add some new features. How can we open this compilation file, and which programs can we use to edit this file?