Kl0ibi/esp32_hackingtool #20: Cant build project in espressif-IDE (iram overflow)

Playedswag, 2024-01-04:
I cant build the project and flash it, i get this error:
 c:/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: hackingtool.elf section `.iram0.text' will not fit in region `iram0_0_seg'
c:/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: IRAM0 segment data does not fit.
c:/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: region `iram0_0_seg' overflowed by 2416 bytes
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
Build complete (3 errors, 0 warnings): C:\esp32_hackingtool-main\build
C:\Espressif\python_env\idf4.4_py3.11_env\Scripts\python.exe C:\Espressif\frameworks\esp-idf-v4.4.6\tools\idf_size.py C:/esp32_hackingtool-main/build/hackingtool.map
Total sizes:

Used static DRAM:   55156 bytes (  69424 remain, 44.3% used)

      .data size:   20052 bytes

      .bss  size:   35104 bytes

Used static IRAM:  133486 bytes (  -2414 remain, 101.8% used) Overflow detected! You can run idf.py size-files for more information.

      .text size:  132459 bytes

   .vectors size:    1027 bytes

Used Flash size : 2135030 bytes

      .text     : 1017439 bytes

      .rodata   : 1117335 bytes

Total image size: 2288568 bytes (.bin may be padded larger). Can i lower the usage somehow?

Kl0ibi, 2024-01-04:
Hey, please lower the log level especially for esp-idf files like bluetooth and wifi!

Playedswag, 2024-01-04:
> Hey, please lower the log level especially for esp-idf files like bluetooth and wifi!

Im not sure but i think, i have not enabled logging for bluetooth and i cant find the option for wifi.

M6ct0night, 2024-08-08:
We want to extract the files from the compilation, change its language, and add some new features. How can we open this compilation file, and which programs can we use to edit this file?