Kl0ibi/esp32_hackingtool #33: How to run stuff on cheap yellow screen?

Trhug25, 2024-06-24:
Hello. I have a cheap gold screen (cyd) and wonder how to install this tool on a cheap gold screen (cyd). I tried installing it but it doesn't show up, it just flashes a green light.If you have a way please tell me. Thank you 
[image]
[image]

InfoDigitalEL, 2024-06-26:
you need to define the display pins in the code, I also wanted to know how to do this using esp-idf menuconfig but I didn't find the options to make changes to the code, and unfortunately the code owner doesn't respond to comments.

Trhug25, 2024-06-27:
You can use gpt chat to edit. I'm trying that too but can't determine the code for the display pin yet. If you identify it, please share it with me. Thank

InfoDigitalEL, 2024-06-27:
I tried help with chatgpt but without success.
What version of esp-idf are you using?

Kl0ibi, 2024-06-27:
hi, i think you are looking for line 1784 - 1792 in my provided sdkconfig.h file.
```
CONFIG_MIPI_DISPLAY_SPI_HOST=0x01
CONFIG_MIPI_DISPLAY_PIN_MISO=-1
CONFIG_MIPI_DISPLAY_PIN_MOSI=19
CONFIG_MIPI_DISPLAY_PIN_CLK=18
CONFIG_MIPI_DISPLAY_PIN_CS=5
CONFIG_MIPI_DISPLAY_PIN_DC=16
CONFIG_MIPI_DISPLAY_PIN_RST=23
CONFIG_MIPI_DISPLAY_PIN_BL=4
CONFIG_MIPI_DISPLAY_PWM_BL=-1
```
note: you can search in the menuconfig with vim controls --> just type / and what you are searching for, example /DISPLAY_PIN and you should find it.

btw I am using idf version 4.3.2

hope I could help

InfoDigitalEL, 2024-06-27:
I'm having some difficulties compiling the project, I would like to do some translations including the capitive portal pages but I have some build errors, could you help me? I will pay for it. my email infozapbot@gmail.com

Trhug25, 2024-06-27:
Did you find cheap gold monitor display pins? I have not found

Trhug25, 2024-06-28:
I can not do it. Someone please help me :(

TechsavvyCNC, 2024-08-13:
Did anybody figure out how to use this screen and the exact pinout?