Kl0ibi/esp32_hackingtool #38: Esp idf component error

guavuavcycayfayqvqu, 2024-07-14:
No matter how i try add in cmake/dependency the error never goes. Added in photo. Sorry for dump question I'm noob. I'm building   for esp32 s3 
[image]

guavuavcycayfayqvqu, 2024-07-14:
https://esp32.com/viewtopic.php?f=2&t=40881

Kl0ibi, 2024-07-14:
Hey, you are facing the issue on a repo I am not the owner of. But I'm assuming you simply installed the wrong ESP-IDF version.

closed 2024-07-14 by Kl0ibi

guavuavcycayfayqvqu, 2024-07-15:
> Hey, you are facing the issue on a repo I am not the owner of. But I'm assuming you simply installed the wrong ESP-IDF version.

I used latest version.

Kl0ibi, 2024-07-15:
Please use the IDF Version as described in each description: For my project use 4.3.2 and for the other one use 4.1

guavuavcycayfayqvqu, 2024-07-15:
> Please use the IDF Version as described in each description: For my project use 4.3.2 and for the other one use 4.1

Okay i will try and give update

guavuavcycayfayqvqu, 2024-07-15:
[image]
Not working

guavuavcycayfayqvqu, 2024-07-15:
> Please use the IDF Version as described in each description: For my project use 4.3.2 and for the other one use 4.1

Can you please build for esp32 s3 and upload in github / send in my mail? I really liked your project. Please. (I can help in debug and testing in esp32 s3)

M6ct0night, 2024-08-08:
We want to extract the files from the compilation, change its language, and add some new features. How can we open this compilation file, and which programs can we use to edit this file?