Kl0ibi/esp32_hackingtool #45: Add new cportal

mehditoxic, 2024-08-13:
how can i add customised cportal?

kilarhasina6, 2024-08-21:
> how can i add customised cportal? @Kl0ibi

Hahah joker bro🤡.
Just modify html css. Why open issues for question? This is no question 😠

mehditoxic, 2024-08-21:
> > how can i add customised cportal? @Kl0ibi
> 
> Hahah joker bro🤡.
> Just modify html css. Why open issues for question? This is no question 😠

It’s not just about the web page. The names of the access points also need to be updated. My main question is how to add customized options to the items?

kilarhasina6, 2024-08-21:
Why will you need to change access point,? 😂

mehditoxic, 2024-08-21:
> Why will you need to change access point,? 😂
> 
> 
> > how can i add customised cportal? @Kl0ibi <https://github.com/Kl0ibi>
> >
> > Hahah joker bro🤡.
> > Just modify html css. Why open issues for question? This is no question 😠
> >
> > It’s not just about the web page. The names of the access points also need
> > to be updated. My main question is how to add customized options to the
> > items?

Don't you expect me to run the personalized page with the name 'Free Google WiFi'?

Kl0ibi, 2024-08-21:
Hi, htool_wifi.c check line 642 and above for SSID and line 255 for html file

closed 2024-08-21 by Kl0ibi